pqpo/SmartCamera #9: 小米mix2手机上，拍照出来的结果与预期不相符

matbird, 2018-09-22:
拍出来的照片跟预期的向左旋转了90度，
在华为麦芒的手机上测试是正常的

matbird, 2018-09-22:
mCameraView.addCallback(new CameraView.Callback() {
            @Override
            public void onPictureTaken(CameraView cameraView, byte[] data) {
                super.onPictureTaken(cameraView, data);
               // 发现这个地方的data出来就已经旋转了
                image2.setImageBitmap(BitmapFactory.decodeByteArray(data, 0, data.length));
                mCameraView.cropImage(data, new SmartCameraView.CropCallback() {
                    @Override
                    public void onCropped(Bitmap bitmap) {
                        if(bitmap != null){
//                            testImageView.setImageBitmap(bitmap);
                            mPreview.setImageBitmap(bitmap);
                            String saveString = Utils.saveBitmap(bitmap);
                            showTip(saveString);
                            mCameraView.stopScan();
                        }
                    }
                });
            }
        });

pqpo, 2018-09-27:
下个版本我更新一下，如果你解决了也可以给我提个 merge。

pqpo, 2018-09-27:
https://github.com/pqpo/SmartCamera/issues/8

jinke1984, 2018-09-29:
在三星的手机里面也发现了这种情况，我修改了下cropImage这个方法就可以了
public void cropImage(final byte[] data, final CropCallback cropCallback) {
        new Thread() {
            @Override
            public void run() {
                super.run();
                int degress = 0;
                Rect revisedMaskRect = getAdjustPictureMaskRect();
                try{
                    final ExifInterface exifInterface = new ExifInterface(FormatTools.getInstance().Byte2InputStream(data));
                    final int attributeInt = exifInterface.getAttributeInt(ExifInterface.TAG_ORIENTATION,
                            ExifInterface.ORIENTATION_UNDEFINED);
                    switch (attributeInt) {
                        case ExifInterface.ORIENTATION_ROTATE_90:
                            Log.e("jinke","90");
                            degress = 90;
                            break;
                        case ExifInterface.ORIENTATION_ROTATE_180:
                            Log.e("jinke","180");
                            degress = 180;
                            break;
                        case ExifInterface.ORIENTATION_ROTATE_270:
                            Log.e("jinke","270");
                            degress = 270;
                            break;
                        default:
                            break;
                    }
                }catch (IOException e){
                    e.printStackTrace();
                }
                final Matrix matrix = new Matrix();
                matrix.setRotate(degress);
                Bitmap tempBit = BitmapFactory.decodeByteArray(data, 0, data.length);
                Bitmap bitmapSrc = Bitmap.createBitmap(tempBit, 0, 0, tempBit.getWidth(), tempBit.getHeight(), matrix, true);
                if (revisedMaskRect != null) {
                    final Bitmap bitmap = Bitmap.createBitmap(bitmapSrc, revisedMaskRect.left,
                            revisedMaskRect.top, revisedMaskRect.width(), revisedMaskRect.height());
                    bitmapSrc.recycle();
                    tempBit.recycle();
                    post(new Runnable() {
                        @Override
                        public void run() {
                            cropCallback.onCropped(bitmap);
                        }
                    });
                    return;
                }
                post(new Runnable() {
                    @Override
                    public void run() {
                        cropCallback.onCropped(null);
                    }
                });
            }
        }.start();
    }

pqpo, 2018-09-29:
使用的是 com.android.support:exifinterface ？

pqpo, 2018-09-29:
v1.3.0 版本已解决

closed 2018-09-29 by pqpo