pqpo/SmartCamera #14: 启动崩溃

ljdang, 2018-09-30:
无论是用提供的apk还是代码编译，都出现启动崩溃，换了不同手机同样出现。
log如下：
09-30 16:50:01.903 25961-25961/me.pqpo.smartcamera E/libc: Access denied finding property "camera.hal1.packagelist"
09-30 16:50:02.032 25961-25961/me.pqpo.smartcamera E/libc: Access denied finding property "camera.hal1.packagelist"
09-30 16:50:02.935 25961-26018/me.pqpo.smartcamera E/Camera-JNI: Callback buffer was too small! Expected 4147200 bytes, but got 28800 bytes!
    Couldn't allocate byte array for JPEG data
09-30 16:50:02.948 25961-26264/me.pqpo.smartcamera E/cv::error(): OpenCV Error: Assertion failed (ssize.area() > 0) in void cv::resize(cv::InputArray, cv::OutputArray, cv::Size, double, double, int), file /hdd2/buildbot/slaves/slave_ardbeg1/50-SDK/opencv/modules/imgproc/src/imgwarp.cpp, line 1968
09-30 16:50:02.949 25961-26264/me.pqpo.smartcamera A/libc: Fatal signal 6 (SIGABRT), code -6 in tid 26264 (processThread), pid 25961 (qpo.smartcamera)

pqpo, 2018-09-30:
机型，系统版本提供一下

ljdang, 2018-09-30:
@pqpo 都是魅族的机型，魅族16，Android版本8.1.0；魅族M15， Android版本7.1.2

sureshkumarece, 2018-10-16:
First Thanks  for the great lib.
The same Error for me also

OS VERSION: 5.1.1
DEVICE :SONY SGP321

10-16 16:05:16.335 2318-2447/me.pqpo.smartcamera E/cv::error(): OpenCV Error: Assertion failed (ssize.area() > 0) in void cv::resize(cv::InputArray, cv::OutputArray, cv::Size, double, double, int), file /hdd2/buildbot/slaves/slave_ardbeg1/50-SDK/opencv/modules/imgproc/src/imgwarp.cpp, line 1968
    
    --------- beginning of crash
10-16 16:05:16.335 2318-2447/me.pqpo.smartcamera A/libc: Fatal signal 6 (SIGABRT), code -6 in tid 2447 (processThread)
10-16 16:05:16.341 410-2439/? E/mm-camera: axi_command_ops cmd_type = 2
10-16 16:05:16.365 410-2439/? E/mm-camera: axi_proc_unregister_wms: Releasing write master 0
10-16 16:05:16.365 410-2439/? E/mm-camera: axi_proc_unregister_wms: Releasing write master 1
10-16 16:05:16.365 410-2439/? E/mm-camera: axi_proc_unregister_wms: Releasing write master 2
10-16 16:05:16.365 410-2439/? E/mm-camera: axi_proc_unregister_wms: Releasing write master 3
10-16 16:05:16.365 410-2439/? E/mm-camera: axi_proc_unregister_wms: Releasing write master 4
10-16 16:05:16.365 410-2439/? E/mm-camera: axi_proc_unregister_wms: Releasing write master 5
10-16 16:05:16.365 410-2439/? E/mm-camera: axi_proc_unregister_wms: Releasing write master 6
10-16 16:05:16.365 410-2439/? E/mm-camera: axi_command_ops cmd_type = 3
10-16 16:05:16.393 410-2439/? E/mm-camera: axi_set_params: begin, axi_obj_idx = 0
10-16 16:05:16.393 410-2439/? E/mm-camera: axi_set_params: begin, axi_obj_idx = 0
10-16 16:05:16.393 410-2439/? E/mm-camera: axi_set_params: begin, axi_obj_idx = 0
10-16 16:05:16.393 410-2439/? E/mm-camera: axi_set_params: begin, axi_obj_idx = 0
10-16 16:05:16.395 410-2439/? E/mm-camera: axi_command_ops cmd_type = 0
10-16 16:05:16.422 410-2439/? E/mm-camera: axi_set_params: begin, axi_obj_idx = 0
10-16 16:05:16.422 410-2439/? E/mm-camera: axi_set_params: begin, axi_obj_idx = 0
10-16 16:05:16.422 410-2439/? E/mm-camera: axi_set_params: begin, axi_obj_idx = 0
10-16 16:05:16.422 410-2439/? E/mm-camera: axi_config: begin
10-16 16:05:16.422 410-2439/? E/mm-camera: axi_vfe_config: begin
10-16 16:05:16.422 410-2439/? E/mm-camera: current target is not 8930
10-16 16:05:16.423 410-2439/? E/mm-camera: axi_vfe_config: Primary Y, wm = 0!!
10-16 16:05:16.423 410-2439/? E/mm-camera: axi_vfe_config: Primary CbCr, wm = 1!!
10-16 16:05:16.423 410-2439/? E/mm-camera: axi_vfe_config:Primary image format 1
10-16 16:05:16.423 410-2439/? E/mm-camera: axi_vfe_config: AXI Config
10-16 16:05:16.423 410-2439/? E/mm-camera: AXI Output busCmd = 0
10-16 16:05:16.423 410-2439/? E/mm-camera: AXI Output busCfg = 44738417
10-16 16:05:16.423 410-2439/? E/mm-camera: AXI Output xbarCfg0 = 512
10-16 16:05:16.423 410-2439/? E/mm-camera: AXI Output xbarCfg1 = 0
10-16 16:05:16.423 410-2439/? E/mm-camera: AXI Output busWrSkipCfg = 0
10-16 16:05:16.423 410-2439/? E/mm-camera: configure_wm_registers: axi channel write master 0 config
10-16 16:05:16.423 410-2439/? E/mm-camera: AXI Output busUbOffset = 0
10-16 16:05:16.423 410-2439/? E/mm-camera: AXI Output busUbDepth = 783
10-16 16:05:16.423 410-2439/? E/mm-camera: AXI Output buslinesPerImage = 1199
10-16 16:05:16.423 410-2439/? E/mm-camera: AXI Output busdwordsPerLine = 99
10-16 16:05:16.423 410-2439/? E/mm-camera: AXI Output busburstLength = 1
10-16 16:05:16.423 410-2439/? E/mm-camera: AXI Output busbufferNumRows = 1199
10-16 16:05:16.423 410-2439/? E/mm-camera: AXI Output busrowIncrement = 200
10-16 16:05:16.423 410-2439/? E/mm-camera: configure_wm_registers: axi channel write master 1 config
10-16 16:05:16.423 410-2439/? E/mm-camera: AXI Output busUbOffset = 784
10-16 16:05:16.423 410-2439/? E/mm-camera: AXI Output busUbDepth = 135
10-16 16:05:16.423 410-2439/? E/mm-camera: AXI Output buslinesPerImage = 599
10-16 16:05:16.423 410-2439/? E/mm-camera: AXI Output busdwordsPerLine = 99
10-16 16:05:16.423 410-2439/? E/mm-camera: AXI Output busburstLength = 1
10-16 16:05:16.423 410-2439/? E/mm-camera: AXI Output busbufferNumRows = 599
10-16 16:05:16.423 410-2439/? E/mm-camera: AXI Output busrowIncrement = 200
10-16 16:05:16.423 410-2439/? E/mm-camera: axi_config AXI Configuration success 
10-16 16:05:16.423 410-2439/? E/mm-camera: axi_command_ops cmd_type = 1
10-16 16:05:16.460 410-2439/? E/mm-camera: axi_proc_reg_update: write master 0 is active
10-16 16:05:16.460 410-2439/? E/mm-camera: axi_proc_reg_update: write master 1 is active
10-16 16:05:16.587 1141-1141/? E/fast-dormancy: [FDM]:  rrc_state=0
10-16 16:05:17.043 1141-1141/? E/fast-dormancy: [FDM]:  rrc_state=0
10-16 16:05:17.291 1141-1141/? E/fast-dormancy: [FDM]:  rrc_state=0
10-16 16:05:17.383 772-1223/? E/InputDispatcher: channel '3b933cc0 me.pqpo.smartcamera/me.pqpo.smartcamera.MainActivity (server)' ~ Channel is unrecoverably broken and will be disposed!
10-16 16:05:17.390 410-2439/? E/mm-camera: axi_command_ops cmd_type = 2
10-16 16:05:17.403 410-2439/? E/mm-camera: axi_proc_unregister_wms: Releasing write master 0
10-16 16:05:17.403 410-2439/? E/mm-camera: axi_proc_unregister_wms: Releasing write master 1
10-16 16:05:17.403 410-2439/? E/mm-camera: axi_proc_unregister_wms: Releasing write master 2
10-16 16:05:17.403 410-2439/? E/mm-camera: axi_proc_unregister_wms: Releasing write master 3
10-16 16:05:17.403 410-2439/? E/mm-camera: axi_proc_unregister_wms: Releasing write master 4
10-16 16:05:17.403 410-2439/? E/mm-camera: axi_proc_unregister_wms: Releasing write master 5
10-16 16:05:17.403 410-2439/? E/mm-camera: axi_proc_unregister_wms: Releasing write master 6
10-16 16:05:17.403 410-2439/? E/mm-camera: axi_command_ops cmd_type = 3

pqpo, 2018-10-16:
I can't reproduce it. and I upgraded a version v1.3.1. Can you give it a try?

sureshkumarece, 2018-10-17:
As per my debugging the issue occurred due to maskRect  height and width.So I modified the code as below.Its now working like charm.Thanks

public int previewScan(byte[] yuvData, int width, int height, int rotation, Rect maskRect) {
if(maskRect.height()<=0||maskRect.width()<=0)
            return -1;
........