pqpo/SmartCamera #24: arm64-v8a架构编译不了

panhongjin, 2018-12-10:
在opencv\lib\arm64-v8a目录下添加OpenCV 3.4.4版libopencv_core、libopencv_imgproc和libtbb库，编译还是报错。
Execution failed for task ':smartcameralib:externalNativeBuildDebug'.
> Build command failed.
  Error while executing process D:\Android\android-sdk\cmake\3.6.4111459\bin\cmake.exe with arguments {--build D:\workspace\SmartCamera-master\smartcameralib\.externalNativeBuild\cmake\debug\arm64-v8a --target smart_camera}
  [1/1] Linking CXX shared library ..\..\..\..\build\intermediates\cmake\debug\obj\arm64-v8a\libsmart_camera.so
  FAILED: cmd.exe /C "cd . && D:\Android\android-sdk\ndk-bundle\toolchains\llvm\prebuilt\windows-x86_64\bin\clang++.exe  --target=aarch64-none-linux-android --gcc-toolchain=D:/Android/android-sdk/ndk-bundle/toolchains/aarch64-linux-android-4.9/prebuilt/windows-x86_64 --sysroot=D:/Android/android-sdk/ndk-bundle/sysroot -fPIC -isystem D:/Android/android-sdk/ndk-bundle/sysroot/usr/include/aarch64-linux-android -D__ANDROID_API__=21 -g -DANDROID -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes -Wa,--noexecstack -Wformat -Werror=format-security  -std=c++11 -frtti -lz -O0 -fno-limit-debug-info  -Wl,--exclude-libs,libgcc.a -Wl,--exclude-libs,libatomic.a --sysroot D:/Android/android-sdk/ndk-bundle/platforms/android-21/arch-arm64 -Wl,--build-id -Wl,--warn-shared-textrel -Wl,--fatal-warnings -Wl,--no-undefined -Wl,-z,noexecstack -Qunused-arguments -Wl,-z,relro -Wl,-z,now -shared -Wl,-soname,libsmart_camera.so -o ..\..\..\..\build\intermediates\cmake\debug\obj\arm64-v8a\libsmart_camera.so CMakeFiles/smart_camera.dir/src/main/cpp/android_utils.cpp.o CMakeFiles/smart_camera.dir/src/main/cpp/opencv_utils.cpp.o CMakeFiles/smart_camera.dir/src/main/cpp/smart_camera.cpp.o  ../../../../opencv/lib/arm64-v8a/libopencv_imgproc.a ../../../../opencv/lib/arm64-v8a/libopencv_core.a -llog -ljnigraphics -latomic -lm "D:/Android/android-sdk/ndk-bundle/sources/cxx-stl/gnu-libstdc++/4.9/libs/arm64-v8a/libsupc++.a" "D:/Android/android-sdk/ndk-bundle/sources/cxx-stl/gnu-libstdc++/4.9/libs/arm64-v8a/libgnustl_shared.so" && cd ."
  ../../../../opencv/lib/arm64-v8a/libopencv_core.a(parallel.cpp.o): In function `tbb::task_scheduler_init::~task_scheduler_init()':
  parallel.cpp:(.text._ZN3tbb19task_scheduler_initD2Ev[_ZN3tbb19task_scheduler_initD5Ev]+0x8): undefined reference to `tbb::task_scheduler_init::terminate()'
  ../../../../opencv/lib/arm64-v8a/libopencv_core.a(parallel.cpp.o): In function `tbb::interface7::internal::start_for<tbb::blocked_range<int>, (anonymous namespace)::ProxyLoopBody, tbb::auto_partitioner const>::execute()':
  parallel.cpp:(.text._ZN3tbb10interface78internal9start_forINS_13blocked_rangeIiEEN12_GLOBAL__N_113ProxyLoopBodyEKNS_16auto_partitionerEE7executeEv+0x84): undefined reference to `tbb::internal::allocate_continuation_proxy::allocate(unsigned long) const'
  parallel.cpp:(.text._ZN3tbb10interface78internal9start_forINS_13blocked_rangeIiEEN12_GLOBAL__N_113ProxyLoopBodyEKNS_16auto_partitionerEE7executeEv+0xac): undefined reference to `tbb::internal::allocate_child_proxy::allocate(unsigned long) const'
  parallel.cpp:(.text._ZN3tbb10interface78internal9start_forINS_13blocked_rangeIiEEN12_GLOBAL__N_113ProxyLoopBodyEKNS_16auto_partitionerEE7executeEv+0x2e4): undefined reference to `tbb::internal::allocate_continuation_proxy::allocate(unsigned long) const'
  parallel.cpp:(.text._ZN3tbb10interface78internal9start_forINS_13blocked_rangeIiEEN12_GLOBAL__N_113ProxyLoopBodyEKNS_16auto_partitionerEE7executeEv+0x320): undefined reference to `tbb::internal::allocate_child_proxy::allocate(unsigned long) const'
  parallel.cpp:(.text._ZN3tbb10interface78internal9start_forINS_13blocked_rangeIiEEN12_GLOBAL__N_113ProxyLoopBodyEKNS_16auto_partitionerEE7executeEv+0x3f0): undefined reference to `tbb::task_group_context::is_group_execution_cancelled() const'
  ../../../../opencv/lib/arm64-v8a/libopencv_core.a(parallel.cpp.o): In function `cv::parallel_for_(cv::Range const&, cv::ParallelLoopBody const&, double)':
  parallel.cpp:(.text._ZN2cv13parallel_for_ERKNS_5RangeERKNS_16ParallelLoopBodyEd+0xc4): undefined reference to `tbb::task_group_context::init()'
  parallel.cpp:(.text._ZN2cv13parallel_for_ERKNS_5RangeERKNS_16ParallelLoopBodyEd+0xd4): undefined reference to `tbb::internal::allocate_root_with_context_proxy::allocate(unsigned long) const'
  parallel.cpp:(.text._ZN2cv13parallel_for_ERKNS_5RangeERKNS_16ParallelLoopBodyEd+0x11c): undefined reference to `tbb::internal::get_initial_auto_partitioner_divisor()'
  parallel.cpp:(.text._ZN2cv13parallel_for_ERKNS_5RangeERKNS_16ParallelLoopBodyEd+0x14c): undefined reference to `tbb::task_group_context::~task_group_context()'
  parallel.cpp:(.text._ZN2cv13parallel_for_ERKNS_5RangeERKNS_16ParallelLoopBodyEd+0x170): undefined reference to `tbb::task_group_context::~task_group_context()'
  parallel.cpp:(.text._ZN2cv13parallel_for_ERKNS_5RangeERKNS_16ParallelLoopBodyEd+0x188): undefined reference to `tbb::internal::allocate_root_with_context_proxy::free(tbb::task&) const'
  ../../../../opencv/lib/arm64-v8a/libopencv_core.a(parallel.cpp.o): In function `cv::getNumThreads()':
  parallel.cpp:(.text._ZN2cv13getNumThreadsEv+0x24): undefined reference to `tbb::task_scheduler_init::default_num_threads()'
  ../../../../opencv/lib/arm64-v8a/libopencv_core.a(parallel.cpp.o): In function `cv::setNumThreads(int)':
  parallel.cpp:(.text._ZN2cv13setNumThreadsEi+0x28): undefined reference to `tbb::task_scheduler_init::terminate()'
  parallel.cpp:(.text._ZN2cv13setNumThreadsEi+0x44): undefined reference to `tbb::task_scheduler_init::initialize(int)'
  ../../../../opencv/lib/arm64-v8a/libopencv_core.a(parallel.cpp.o): In function `_GLOBAL__sub_I_parallel.cpp':
  parallel.cpp:(.text.startup._GLOBAL__sub_I_parallel.cpp+0x20): undefined reference to `tbb::task_scheduler_init::initialize(int, unsigned long)'
  ../../../../opencv/lib/arm64-v8a/libopencv_core.a(parallel.cpp.o):(.data.rel.ro._ZTIN3tbb10interface78internal9flag_taskE[_ZTIN3tbb10interface78internal9flag_taskE]+0x10): undefined reference to `typeinfo for tbb::task'
  ../../../../opencv/lib/arm64-v8a/libopencv_core.a(parallel.cpp.o):(.data.rel.ro._ZTVN3tbb10interface78internal9flag_taskE[_ZTVN3tbb10interface78internal9flag_taskE]+0x28): undefined reference to `tbb::task::note_affinity(unsigned short)'
  ../../../../opencv/lib/arm64-v8a/libopencv_core.a(parallel.cpp.o):(.data.rel.ro._ZTIN3tbb10interface78internal9start_forINS_13blocked_rangeIiEEN12_GLOBAL__N_113ProxyLoopBodyEKNS_16auto_partitionerEEE+0x10): undefined reference to `typeinfo for tbb::task'
  clang++.exe: error: linker command failed with exit code 1 (use -v to see invocation)
  ninja: build stopped: subcommand failed.

pqpo, 2018-12-10:
OpenCv 版本号 ：v2.4.11

panhongjin, 2018-12-11:
OpenCV v2.4.11版没有arm64-v8a架构的，我使用OpenCV v3.4.4版的arm64-v8a架构，编译报错。

panhongjin, 2018-12-11:
OpenCV v2.4.11版没有arm64-v8a架构的，我使用OpenCV v3.4.4版的arm64-v8a架构，编译报错。
工程在努比亚Z18mini  8.1.0机型上适配有问题，该机型是arm64-v8a(AArch64)架构的。

pqpo, 2019-07-11:
使用最新版本：v2.0.0 
https://github.com/pqpo/SmartCamera/releases

closed 2019-07-11 by pqpo

panhongjin, 2019-10-27:
> 使用最新版本：v2.0.0
> https://github.com/pqpo/SmartCamera/releases

非常感谢