pqpo/SmartCamera #25: 启动崩溃怎么回事

GitHub-xiaomili, 2018-12-20:
12-20 09:45:00.821 18489-18489/cn.fundview.agriculturemarket.debug E/AndroidRuntime: FATAL EXCEPTION: main
                                                                                     Process: cn.fundview.agriculturemarket.debug, PID: 18489
                                                                                     java.lang.RuntimeException: Unable to start activity ComponentInfo{cn.fundview.agriculturemarket.debug/cn.fundview.agriculturemarket.mvp.features.test.TestActivity}: android.view.InflateException: Binary XML file line #6: Binary XML file line #6: Error inflating class me.pqpo.smartcameralib.SmartCameraView
                                                                                         at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2915)
                                                                                         at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2980)
                                                                                         at android.app.ActivityThread.-wrap12(ActivityThread.java)
                                                                                         at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1686)
                                                                                         at android.os.Handler.dispatchMessage(Handler.java:110)
                                                                                         at android.os.Looper.loop(Looper.java:203)
                                                                                         at android.app.ActivityThread.main(ActivityThread.java:6533)
                                                                                         at java.lang.reflect.Method.invoke(Native Method)
                                                                                         at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1113)
                                                                                         at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:974)
                                                                                      Caused by: android.view.InflateException: Binary XML file line #6: Binary XML file line #6: Error inflating class me.pqpo.smartcameralib.SmartCameraView
                                                                                      Caused by: android.view.InflateException: Binary XML file line #6: Error inflating class me.pqpo.smartcameralib.SmartCameraView
                                                                                      Caused by: java.lang.reflect.InvocationTargetException
                                                                                         at java.lang.reflect.Constructor.newInstance0(Native Method)
                                                                                         at java.lang.reflect.Constructor.newInstance(Constructor.java:430)
                                                                                         at android.view.LayoutInflater.createView(LayoutInflater.java:645)
                                                                                         at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:787)
                                                                                         at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:727)
                                                                                         at android.view.LayoutInflater.rInflate(LayoutInflater.java:858)
                                                                                         at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:821)
                                                                                         at android.view.LayoutInflater.inflate(LayoutInflater.java:518)
                                                                                         at android.view.LayoutInflater.inflate(LayoutInflater.java:426)
                                                                                         at android.view.LayoutInflater.inflate(LayoutInflater.java:377)
                                                                                         at android.support.v7.app.AppCompatDelegateImplV9.setContentView(AppCompatDelegateImplV9.java:287)
                                                                                         at android.support.v7.app.AppCompatActivity.setContentView(AppCompatActivity.java:139)
                                                                                         at cn.fundview.agriculturemarket.mvp.features.test.TestActivity.onCreate(TestActivity.java:30)
                                                                                         at android.app.Activity.performCreate(Activity.java:6710)
                                                                                         at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1122)
                                                                                         at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2868)
                                                                                         at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2980)
                                                                                         at android.app.ActivityThread.-wrap12(ActivityThread.java)
                                                                                         at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1686)
                                                                                         at android.os.Handler.dispatchMessage(Handler.java:110)
                                                                                         at android.os.Looper.loop(Looper.java:203)
                                                                                         at android.app.ActivityThread.main(ActivityThread.java:6533)
                                                                                         at java.lang.reflect.Method.invoke(Native Method)
                                                                                         at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1113)
                                                                                         at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:974)
                                                                                      Caused by: android.view.InflateException: Binary XML file line #16: Binary XML file line #16: Error inflating class android.view.TextureView
                                                                                      Caused by: android.view.InflateException: Binary XML file line #16: Error inflating class android.view.TextureView
                                                                                      Caused by: java.lang.reflect.InvocationTargetException
                                                                                         at java.lang.reflect.Constructor.newInstance0(Native Method)
                                                                                         at java.lang.reflect.Constructor.newInstance(Constructor.java:430)
                                                                                         at android.view.LayoutInflater.createView(LayoutInflater.java:645)
                                                                                         at android.view.LayoutInflater.onCreateView(LayoutInflater.java:700)
                                                                                         at com.android.internal.policy.PhoneLayoutInflater.onCreateView(PhoneLayoutInflater.java:68)
                                                                                         at android.view.LayoutInflater.onCreateView(LayoutInflater.java:717)
                                                                                         at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:785)
[... cut ...]
                                                                                         at android.view.View.<init>(View.java:4170)
                                                                                         at android.view.View.<init>(View.java:4149)
                                                                                         at android.view.TextureView.<init>(TextureView.java:151)
                                                                                         	... 42 more

geekqian, 2019-02-18:
看下这个文章
https://blog.csdn.net/geekqian/article/details/87659786

pqpo, 2019-07-11:
试试 v2.0.1

closed 2019-07-11 by pqpo