pqpo/SmartCamera #26: couldn’t find “libsmart_camera.so”

GitHub-xiaomili, 2018-12-20:
Process: cn.fundview.smartcropperdemo, PID: 24063
                                                                              java.lang.RuntimeException: Unable to start activity ComponentInfo{cn.fundview.smartcropperdemo/cn.fundview.smartcropperdemo.MainActivity}: android.view.InflateException: Binary XML file line #6: Binary XML file line #6: Error inflating class me.pqpo.smartcameralib.SmartCameraView
                                                                                  at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2915)
                                                                                  at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2980)
                                                                                  at android.app.ActivityThread.-wrap12(ActivityThread.java)
                                                                                  at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1686)
                                                                                  at android.os.Handler.dispatchMessage(Handler.java:110)
                                                                                  at android.os.Looper.loop(Looper.java:203)
                                                                                  at android.app.ActivityThread.main(ActivityThread.java:6533)
                                                                                  at java.lang.reflect.Method.invoke(Native Method)
                                                                                  at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1113)
                                                                                  at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:974)
                                                                               Caused by: android.view.InflateException: Binary XML file line #6: Binary XML file line #6: Error inflating class me.pqpo.smartcameralib.SmartCameraView
                                                                               Caused by: android.view.InflateException: Binary XML file line #6: Error inflating class me.pqpo.smartcameralib.SmartCameraView
                                                                               Caused by: java.lang.reflect.InvocationTargetException
                                                                                  at java.lang.reflect.Constructor.newInstance0(Native Method)
                                                                                  at java.lang.reflect.Constructor.newInstance(Constructor.java:430)
                                                                                  at android.view.LayoutInflater.createView(LayoutInflater.java:645)
                                                                                  at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:787)
                                                                                  at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:727)
                                                                                  at android.view.LayoutInflater.rInflate(LayoutInflater.java:858)
                                                                                  at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:821)
                                                                                  at android.view.LayoutInflater.inflate(LayoutInflater.java:518)
                                                                                  at android.view.LayoutInflater.inflate(LayoutInflater.java:426)
                                                                                  at android.view.LayoutInflater.inflate(LayoutInflater.java:377)
                                                                                  at android.support.v7.app.AppCompatDelegateImplV9.setContentView(AppCompatDelegateImplV9.java:287)
                                                                                  at android.support.v7.app.AppCompatActivity.setContentView(AppCompatActivity.java:139)
                                                                                  at cn.fundview.smartcropperdemo.MainActivity.onCreate(MainActivity.java:29)
                                                                                  at android.app.Activity.performCreate(Activity.java:6710)
                                                                                  at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1122)
                                                                                  at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2868)
                                                                                  at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2980)
                                                                                  at android.app.ActivityThread.-wrap12(ActivityThread.java)
                                                                                  at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1686)
                                                                                  at android.os.Handler.dispatchMessage(Handler.java:110)
                                                                                  at android.os.Looper.loop(Looper.java:203)
                                                                                  at android.app.ActivityThread.main(ActivityThread.java:6533)
                                                                                  at java.lang.reflect.Method.invoke(Native Method)
                                                                                  at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1113)
                                                                                  at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:974)
12-20 14:49:02.250 24063-24063/cn.fundview.smartcropperdemo E/AndroidRuntime: Caused by: java.lang.UnsatisfiedLinkError: dalvik.system.PathClassLoader[DexPathList[[zip file "/data/app/cn.fundview.smartcropperdemo-2/base.apk", zip file "/data/app/cn.fundview.smartcropperdemo-2/split_lib_dependencies_apk.apk", zip file "/data/app/cn.fundview.smartcropperdemo-2/split_lib_slice_0_apk.apk", zip file "/data/app/cn.fundview.smartcropperdemo-2/split_lib_slice_1_apk.apk", zip file "/data/app/cn.fundview.smartcropperdemo-2/split_lib_slice_2_apk.apk", zip file "/data/app/cn.fundview.smartcropperdemo-2/split_lib_slice_3_apk.apk", zip file "/data/app/cn.fundview.smartcropperdemo-2/split_lib_slice_4_apk.apk", zip file "/data/app/cn.fundview.smartcropperdemo-2/split_lib_slice_5_apk.apk", zip file "/data/app/cn.fundview.smartcropperdemo-2/split_lib_slice_6_apk.apk", zip file "/data/app/cn.fundview.smartcropperdemo-2/split_lib_slice_7_apk.apk", zip file "/data/app/cn.fundview.smartcropperdemo-2/split_lib_slice_8_apk.apk", zip file "/data/app/cn.fundview.smartcropperdemo-2/split_lib_slice_9_apk.apk"],nativeLibraryDirectories=[/data/app/cn.fundview.smartcropperdemo-2/lib/arm64, /data/app/cn.fundview.smartcropperdemo-2/base.apk!/lib/arm64-v8a, /data/app/cn.fundview.smartcropperdemo-2/split_lib_dependencies_apk.apk!/lib/arm64-v8a, /data/app/cn.fundview.smartcropperdemo-2/split_lib_slice_0_apk.apk!/lib/arm64-v8a, /data/app/cn.fundview.smartcropperdemo-2/split_lib_slice_1_apk.apk!/lib/arm64-v8a, /data/app/cn.fundview.smartcropperdemo-2/split_lib_slice_2_apk.apk!/lib/arm64-v8a, /data/app/cn.fundview.smartcropperdemo-2/split_lib_slice_3_apk.apk!/lib/arm64-v8a, /data/app/cn.fundview.smartcropperdemo-2/split_lib_slice_4_apk.apk!/lib/arm64-v8a, /data/app/cn.fundview.smartcropperdemo-2/split_lib_slice_5_apk.apk!/lib/arm64-v8a, /data/app/cn.fundview.smartcropperdemo-2/split_lib_slice_6_apk.apk!/lib/arm64-v8a, /data/app/cn.fundview.smartcropperdemo-2/split_lib_slice_7_apk.apk!/lib/arm64-v8a, /data/app/cn.fundview.smartcropperdemo-2/split_lib_slice_8_apk.apk!/lib/arm64-v8a, /data/app/cn.fundview.smartcropperdemo-2/split_lib_slice_9_apk.apk!/lib/arm64-v8a, /system/lib64, /vendor/lib64]]] couldn't find "libsmart_camera.so"
[... cut ...]
                                                                                  at me.pqpo.smartcameralib.SmartCameraView.init(SmartCameraView.java:50)
                                                                                  at me.pqpo.smartcameralib.SmartCameraView.<init>(SmartCameraView.java:45)
                                                                                  at me.pqpo.smartcameralib.SmartCameraView.<init>(SmartCameraView.java:40)
                                                                                  	... 25 more
下载了github上的项目 里面也没有so库

GitHub-xiaomili, 2018-12-21:
加入so包解决

GitHub-xiaomili, 2018-12-21:
加入so包解决

closed 2018-12-21 by GitHub-xiaomili

USTC-Keyanjie, 2019-07-11:
同问，请问so包在哪里找?@GitHub-xiaomili

geekqian, 2019-07-11:
> 同问，请问so包在哪里找?@GitHub-xiaomili

so包我上传了. 你在这里下载吧. 
链接: https://pan.baidu.com/s/1xczWGcO2MZVo0yl3aVyDLg 提取码: nbek
有需要别的架构你自己去生成. 可以参考我发的这篇博客:
http://www.geekqian.com/post/a6f3e92a.html

pqpo, 2019-07-11:
之前已经将 SmartCropper 的 OpenCV 升级到最新版本了，已经支持最新版本 NDK 编译，并且支持 64 位架构，过几天我会将 SmartCamera 也升级一下。

pqpo, 2019-07-11:
使用最新版本：v2.0.0
https://github.com/pqpo/SmartCamera/releases