pqpo/SmartCamera #30: Android 4.1.2 运行崩溃

hVenus, 2019-03-01:
E/AndroidRuntime: FATAL EXCEPTION: main
    java.lang.RuntimeException: Unable to start activity ComponentInfo{me.pqpo.scan/me.pqpo.scan.MainActivity}: android.view.InflateException: Binary XML file line #10: Error inflating class me.pqpo.smartcameralib.SmartCameraView
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2127)
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2152)
        at android.app.ActivityThread.access$700(ActivityThread.java:144)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1258)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:137)
        at android.app.ActivityThread.main(ActivityThread.java:4944)
        at java.lang.reflect.Method.invokeNative(Native Method)
        at java.lang.reflect.Method.invoke(Method.java:511)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1043)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:810)
        at dalvik.system.NativeStart.main(Native Method)
     Caused by: android.view.InflateException: Binary XML file line #10: Error inflating class me.pqpo.smartcameralib.SmartCameraView
        at android.view.LayoutInflater.createView(LayoutInflater.java:619)
        at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:693)
        at android.view.LayoutInflater.rInflate(LayoutInflater.java:752)
        at android.view.LayoutInflater.inflate(LayoutInflater.java:495)
        at android.view.LayoutInflater.inflate(LayoutInflater.java:397)
        at android.view.LayoutInflater.inflate(LayoutInflater.java:353)
        at android.support.v7.app.AppCompatDelegateImpl.setContentView(AppCompatDelegateImpl.java:469)
        at android.support.v7.app.AppCompatActivity.setContentView(AppCompatActivity.java:140)
        at me.pqpo.scan.MainActivity.onCreate(MainActivity.java:61)
        at android.app.Activity.performCreate(Activity.java:5185)
        at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1094)
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2091)
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2152) 
        at android.app.ActivityThread.access$700(ActivityThread.java:144) 
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1258) 
        at android.os.Handler.dispatchMessage(Handler.java:99) 
        at android.os.Looper.loop(Looper.java:137) 
        at android.app.ActivityThread.main(ActivityThread.java:4944) 
        at java.lang.reflect.Method.invokeNative(Native Method) 
        at java.lang.reflect.Method.invoke(Method.java:511) 
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1043) 
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:810) 
        at dalvik.system.NativeStart.main(Native Method) 
     Caused by: java.lang.reflect.InvocationTargetException
        at java.lang.reflect.Constructor.constructNative(Native Method)
        at java.lang.reflect.Constructor.newInstance(Constructor.java:417)
        at android.view.LayoutInflater.createView(LayoutInflater.java:593)
        at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:693) 
        at android.view.LayoutInflater.rInflate(LayoutInflater.java:752) 
        at android.view.LayoutInflater.inflate(LayoutInflater.java:495) 
        at android.view.LayoutInflater.inflate(LayoutInflater.java:397) 
        at android.view.LayoutInflater.inflate(LayoutInflater.java:353) 
        at android.support.v7.app.AppCompatDelegateImpl.setContentView(AppCompatDelegateImpl.java:469) 
        at android.support.v7.app.AppCompatActivity.setContentView(AppCompatActivity.java:140) 
        at me.pqpo.scan.MainActivity.onCreate(MainActivity.java:61) 
        at android.app.Activity.performCreate(Activity.java:5185) 
        at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1094) 
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2091) 
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2152) 
        at android.app.ActivityThread.access$700(ActivityThread.java:144) 
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1258) 
        at android.os.Handler.dispatchMessage(Handler.java:99) 
        at android.os.Looper.loop(Looper.java:137) 
        at android.app.ActivityThread.main(ActivityThread.java:4944) 
        at java.lang.reflect.Method.invokeNative(Native Method) 
        at java.lang.reflect.Method.invoke(Method.java:511) 
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1043) 
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:810) 
        at dalvik.system.NativeStart.main(Native Method) 
     Caused by: java.lang.ExceptionInInitializerError
        at me.pqpo.smartcameralib.SmartCameraView.init(SmartCameraView.java:50)
        at me.pqpo.smartcameralib.SmartCameraView.<init>(SmartCameraView.java:45)
        at me.pqpo.smartcameralib.SmartCameraView.<init>(SmartCameraView.java:40)
        at java.lang.reflect.Constructor.constructNative(Native Method) 
        at java.lang.reflect.Constructor.newInstance(Constructor.java:417) 
[... 36 lines cut ...]
        at android.view.LayoutInflater.inflate(LayoutInflater.java:397) 
        at android.view.LayoutInflater.inflate(LayoutInflater.java:353) 
        at android.support.v7.app.AppCompatDelegateImpl.setContentView(AppCompatDelegateImpl.java:469) 
        at android.support.v7.app.AppCompatActivity.setContentView(AppCompatActivity.java:140) 
        at me.pqpo.scan.MainActivity.onCreate(MainActivity.java:61) 
        at android.app.Activity.performCreate(Activity.java:5185) 
        at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1094) 
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2091) 
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2152) 
        at android.app.ActivityThread.access$700(ActivityThread.java:144) 
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1258) 
        at android.os.Handler.dispatchMessage(Handler.java:99) 
        at android.os.Looper.loop(Looper.java:137) 
        at android.app.ActivityThread.main(ActivityThread.java:4944) 
        at java.lang.reflect.Method.invokeNative(Native Method) 
        at java.lang.reflect.Method.invoke(Method.java:511) 
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1043) 
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:810) 
        at dalvik.system.NativeStart.main(Native Method) 

没有导入源码，是新建工程，用添加依赖的方式引用你的库。
在 Android 8.0上正常。
在Android 4.1.2  和  4.2 上启动崩溃。报以上错误。

请指教该如何处理.

geekqian, 2019-03-01:
@hVenus 
http://www.geekqian.com/post/a6f3e92a.html

hVenus, 2019-04-13:
@geekqian 用的添加依赖的方法，没法改这个配置啊！！！
非要导入源码吗？

geekqian, 2019-04-13:
> @geekqian 用的添加依赖的方法，没法改这个配置啊！！！
> 非要导入源码吗？

不是的, 我是先用原始工程编译生成了so包, 之后放到我的工程中使用的, 我也跟你一样是添加依赖的方法.

w13897215200, 2019-06-18:
求解啊，我导入了so文件还是报错啊

geekqian, 2019-06-18:
> 求解啊，我导入了so文件还是报错啊

什么样的错误. 贴出来看看

w13897215200, 2019-06-18:
[image]

w13897215200, 2019-06-18:
我是直接把你的so库放到jniLibs里

geekqian, 2019-06-18:
> 我是直接把你的so库放到jniLibs里

[image]
[image]
然后 clean, rebuild, Invalidate and Restart.

w13897215200, 2019-06-18:
好的我试一下！
[image]
这个用删掉吗

geekqian, 2019-06-18:
[image]
[image]
@w13897215200

w13897215200, 2019-06-18:
大神，能给个qq吗，这问题困扰一天了。。万分感谢

w13897215200, 2019-06-18:
[image]
[image]

w13897215200, 2019-06-18:
这样改了之后，我自己项目的一个有道图文识别的库
[image]报错了

w13897215200, 2019-06-18:
[image]

w13897215200, 2019-06-18:
[image]

w13897215200, 2019-06-18:
[image]

w13897215200, 2019-06-18:
> [image]
> [image]
> @w13897215200

geekqian, 2019-06-18:
> 这样改了之后，我自己项目的一个有道图文识别的库
> [image]报错了

把v7a包里的so丢到armeabi, 然后删除v7a, x86两个包

w13897215200, 2019-06-18:
这样操作了还是报错
[image]
[image]

[image]

geekqian, 2019-06-18:
> 这样操作了还是报错
> [image]
> [image]
> 
> [image]

参考下我这个博客, http://www.geekqian.com/post/a6f3e92a.html 自己编译一个armeabi的so文件导进去吧. 可能v7a的包不适用于armeabi吧. 不行再邮件联系我. 有时间帮你看看

pqpo, 2019-07-11:
尝试使用最新版本：v2.0.0
https://github.com/pqpo/SmartCamera/releases

closed 2019-07-11 by pqpo