pqpo/SmartCamera #34: 这个库是不是太大了点

sundaylol, 2019-07-17:
这边不混淆之前打包有44M。。。我真的是惊呆了。真的有这么大么

pqpo, 2019-07-17:
可以考虑不引入 x86 架构动态库或者只引入 32 位或 64 位其中的一个 x86 架构动态库。
x86 架构包占比 90% 以上：
[image]
原因是 OpenCV 的 x86 架构动态库比较大，在官方库基础上已经做了精简，下面是 OpenCV 官方的安卓动态库：
[image]

sundaylol, 2019-07-17:
> 可以考虑不引入 x86 架构动态库或者只引入 32 位或 64 位其中的一个 x86 架构动态库。
> x86 架构包占比 90% 以上：
> [image]
> 原因是 OpenCV 的 x86 架构动态库比较大，在官方库基础上已经做了大量精简，下面是 OpenCV 官方的安卓动态库大小：
> ├── arm64-v8a
> │   └── [17M] libopencv_java4.so
> ├── armeabi-v7a
> │   └── [9.4M] libopencv_java4.so
> ├── x86
> │   └── [45M] libopencv_java4.so
> └── x86_64
> └── [66M] libopencv_java4.so

感谢大大，我这边之前已经把X86和X64的去掉了

closed 2019-07-19 by pqpo