pqpo/SmartCamera #43: 构建源代码的时候报错

douni, 2020-03-26:
* What went wrong:
Execution failed for task ‘:smartcameralib:externalNativeBuildDebug’.
> Build command failed.
Error while executing process D:\android-sdk-windows\cmake\3.10.2.4988404\bin\cmake.exe with arguments {–build D:\AndroidStudioProjects\SmartCamera-master\smartcameralib\.externalNativeBuild\cmake\debug\arm64-v8a –target smart_camera}
[1/1] Linking CXX shared library ..\..\..\..\build\intermediates\cmake\debug\obj\arm64-v8a\libsmart_camera.so
FAILED: ../../../../build/intermediates/cmake/debug/obj/arm64-v8a/libsmart_camera.so
cmd.exe /C “cd . && D:\android-sdk-windows\ndk-bundle\toolchains\llvm\prebuilt\windows-x86_64\bin\clang++.exe –target=aarch64-none-linux-android21 –gcc-toolchain=D:/android-sdk-windows/ndk-bundle/toolchains/llvm/prebuilt/windows-x86_64 –sysroot=D:/android-sdk-windows/ndk-bundle/toolchains/llvm/prebuilt/windows-x86_64/sysroot -fPIC -g -DANDROID -fdata-sections -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes -D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security -std=c++11 -frtti -O0 -fno-limit-debug-info -Wl,–exclude-libs,libgcc_real.a -Wl,–exclude-libs,libatomic.a -static-libstdc++ -Wl,–build-id -Wl,–fatal-warnings -Wl,–no-undefined -Qunused-arguments -shared -Wl,-soname,libsmart_camera.so -o ..\..\..\..\build\intermediates\cmake\debug\obj\arm64-v8a\libsmart_camera.so CMakeFiles/smart_camera.dir/src/main/cpp/android_utils.cpp.o CMakeFiles/smart_camera.dir/src/main/cpp/opencv_utils.cpp.o CMakeFiles/smart_camera.dir/src/main/cpp/smart_camera.cpp.o -LD:/AndroidStudioProjects/SmartCamera-master/smartcameralib/opencv/lib/arm64-v8a -LD:/AndroidStudioProjects/SmartCamera-master/smartcameralib/opencv/3rdparty/arm64-v8a -lopencv_imgproc -lopencv_core D:/android-sdk-windows/ndk-bundle/toolchains/llvm/prebuilt/windows-x86_64/sysroot/usr/lib/aarch64-linux-android/libz.a -ltegra_hal -ltbb -lcpufeatures -llog -ljnigraphics -latomic -lm && cd .”
D:/android-sdk-windows/ndk-bundle/toolchains/llvm/prebuilt/windows-x86_64/sysroot/usr/lib/aarch64-linux-android/libz.a(gzwrite.o): In function `write(int, void const*, unsigned long pass_object_size0)’:
bionic/libc/include/bits/fortify/unistd.h:174: undefined reference to `__write_chk’
bionic/libc/include/bits/fortify/unistd.h:174: undefined reference to `__write_chk’
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.

closed 2020-03-26 by douni

persiona, 2020-09-22:
How to solve it

nxnyk, 2020-10-10:
怎么解决的老哥

gitgeek4dx, 2021-08-18:
使用 ndk19 编译，或者修改cmake文件：

target_link_libraries( # Specifies the target library.
                       ${TARGET}
                       android
                       # Links the target library to the log library
                       # included in the NDK.
                       ${log-lib}
                       jnigraphics
                       z)

if(${ANDROID_ABI} STREQUAL x86 OR ${ANDROID_ABI} STREQUAL x86_64)
target_link_libraries(${TARGET} opencv_imgproc opencv_core ippiw ippicv ittnotify tbb cpufeatures)
else()
target_link_libraries(${TARGET} opencv_imgproc opencv_core tegra_hal tbb cpufeatures)
endif()