pqpo/SmartCamera #45: 可不可以重新build一个release的2.0.0的aar，现在的三十多MB，太大了，之前的也才三点几兆

qq724418408, 2020-07-24:


qq724418408, 2020-07-24:
Execution failed for task ':smartcameralib:externalNativeBuildRelease'.
> Build command failed.
  Error while executing process /xxx/xxx/xxx/android/sdk/cmake/3.6.4111459/bin/cmake with arguments {--build /xxx/xxx/xxx/xxx/SmartCamera/smartcameralib/.externalNativeBuild/cmake/release/arm64-v8a --target smart_camera}
  [1/4] Building CXX object CMakeFiles/smart_camera.dir/src/main/cpp/android_utils.cpp.o
  [2/4] Building CXX object CMakeFiles/smart_camera.dir/src/main/cpp/opencv_utils.cpp.o
  [3/4] Building CXX object CMakeFiles/smart_camera.dir/src/main/cpp/smart_camera.cpp.o
  [4/4] Linking CXX shared library ../../../../build/intermediates/cmake/release/obj/arm64-v8a/libsmart_camera.so
  FAILED: : && /xxx/xxx/xxx/android/sdk/ndk-bundle/toolchains/llvm/prebuilt/darwin-x86_64/bin/clang++  --target=aarch64-none-linux-android23 --gcc-toolchain=/xxx/xxx/xxx/android/sdk/ndk-bundle/toolchains/llvm/prebuilt/darwin-x86_64 --sysroot=/xxx/xxx/xxx/android/sdk/ndk-bundle/toolchains/llvm/prebuilt/darwin-x86_64/sysroot -fPIC -g -DANDROID -fdata-sections -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes -D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security  -std=c++11 -frtti -O2 -DNDEBUG  -Wl,--exclude-libs,libgcc.a -Wl,--exclude-libs,libgcc_real.a -Wl,--exclude-libs,libatomic.a -static-libstdc++ -Wl,--build-id -Wl,--fatal-warnings -Wl,--no-undefined -Qunused-arguments -shared -Wl,-soname,libsmart_camera.so -o ../../../../build/intermediates/cmake/release/obj/arm64-v8a/libsmart_camera.so CMakeFiles/smart_camera.dir/src/main/cpp/android_utils.cpp.o CMakeFiles/smart_camera.dir/src/main/cpp/opencv_utils.cpp.o CMakeFiles/smart_camera.dir/src/main/cpp/smart_camera.cpp.o -L/xxx/xxx/xxx/xxx/SmartCamera/smartcameralib/opencv/lib/arm64-v8a  -L/xxx/xxx/xxx/xxx/SmartCamera/smartcameralib/opencv/3rdparty/arm64-v8a -lopencv_imgproc -lopencv_core /xxx/xxx/xxx/android/sdk/ndk-bundle/toolchains/llvm/prebuilt/darwin-x86_64/sysroot/usr/lib/aarch64-linux-android/libz.a -ltegra_hal -ltbb -lcpufeatures -llog -ljnigraphics -latomic -lm && :
  /xxx/xxx/xxx/android/sdk/ndk-bundle/toolchains/llvm/prebuilt/darwin-x86_64/sysroot/usr/lib/aarch64-linux-android/libz.a(gzwrite.o): In function `write(int, void const*, unsigned long pass_object_size0)':
  bionic/libc/include/bits/fortify/unistd.h:174: undefined reference to `__write_chk'
  bionic/libc/include/bits/fortify/unistd.h:174: undefined reference to `__write_chk'
  clang++: error: linker command failed with exit code 1 (use -v to see invocation)
  ninja: build stopped: subcommand failed.

qq724418408, 2020-07-24:
我自己build的报错信息如上

levianye, 2020-08-26:
我也遇到这个编译问题，这个是咋回事呢

gitgeek4dx, 2021-08-18:
ndk版本大于19会出现这个问题，可以改为ndk19以前编译正常
或者使用高版本ndk的话，改下cmake文件也可以，主要是libz的引用，由findLibrary改为target_link_libraries:

find_library( # Sets the name of the path variable.
              log-lib
              # Specifies the name of the NDK library that
              # you want CMake to locate.
              log )

target_link_libraries( # Specifies the target library.
                       ${TARGET}
                       android
                       # Links the target library to the log library
                       # included in the NDK.
                       ${log-lib}
                       jnigraphics
                       z)

if(${ANDROID_ABI} STREQUAL x86 OR ${ANDROID_ABI} STREQUAL x86_64)
target_link_libraries(${TARGET} opencv_imgproc opencv_core ippiw ippicv ittnotify tbb cpufeatures)
else()
target_link_libraries(${TARGET} opencv_imgproc opencv_core tegra_hal tbb cpufeatures)
endif()