pqpo/SmartCamera #48: 有没有办法让识别的图片尽可能的完美不留白边

qq724418408, 2020-08-10:
需要调整哪些参数可以让识别的证件能尽可能的不出现白边呢？

qq724418408, 2020-08-10:
SmartScanner.cannyThreshold1 = 20;
        SmartScanner.cannyThreshold2 = 50;
        SmartScanner.houghLinesThreshold = 130;
        SmartScanner.houghLinesMinLineLength = 80;
        SmartScanner.houghLinesMaxLineGap = 10;
        SmartScanner.gaussianBlurRadius = 3;
        SmartScanner.detectionRatio = 0.05f;
        SmartScanner.checkMinLengthRatio = 0.5f;
        SmartScanner.maxSize = 300;
        SmartScanner.angleThreshold = 5;

qq724418408, 2020-08-10:
当前的参数设置如上↑

pqpo, 2020-08-10:
SmartScanner.detectionRatio = 0.05f;
识别区域，变小可以减少白边，但是也更难识别了。

closed 2020-08-10 by pqpo

qq724418408, 2020-08-10:
可不可以提供回调返回图片的四个角坐标呢

qq724418408, 2020-08-10:
SmartScanner.detectionRatio = 0.05f;
这个取值最小能取到多少呢？