pqpo/SmartCamera #49: FAILED: ../../../../build/intermediates/cmake/debug/obj/x86_64/libsmart_camera.so

KaiGaofly, 2020-10-13:
你好，看之前的问题，找到一个x86文件夹下的so库，但是这里报错要x86_64下面的so库，我编译通不过了，这个有没有别的办法可以解决的？

Build command failed.
Error while executing process C:\Users\GAOKAI\AppData\Local\Android\Sdk\cmake\3.6.4111459\bin\cmake.exe with arguments {--build D:\APP_android\GitHub\SmartCamera-master\smartcameralib\.cxx\cmake\debug\armeabi-v7a --target smart_camera}
[1/1] Linking CXX shared library ..\..\..\..\build\intermediates\cmake\debug\obj\armeabi-v7a\libsmart_camera.so
FAILED: cmd.exe /C "cd . && C:\Users\GAOKAI\AppData\Local\Android\android-ndk-r21b\toolchains\llvm\prebuilt\windows-x86_64\bin\clang++.exe  --target=armv7-none-linux-androideabi21 --gcc-toolchain=C:/Users/GAOKAI/AppData/Local/Android/android-ndk-r21b/toolchains/llvm/prebuilt/windows-x86_64 --sysroot=C:/Users/GAOKAI/AppData/Local/Android/android-ndk-r21b/toolchains/llvm/prebuilt/windows-x86_64/sysroot -fPIC -g -DANDROID -fdata-sections -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes -D_FORTIFY_SOURCE=2 -march=armv7-a -mthumb -Wformat -Werror=format-security  -std=c++11 -frtti -O0 -fno-limit-debug-info  -Wl,--exclude-libs,libgcc.a -Wl,--exclude-libs,libgcc_real.a -Wl,--exclude-libs,libatomic.a -static-libstdc++ -Wl,--build-id -Wl,--fatal-warnings -Wl,--exclude-libs,libunwind.a -Wl,--no-undefined -Qunused-arguments -shared -Wl,-soname,libsmart_camera.so -o ..\..\..\..\build\intermediates\cmake\debug\obj\armeabi-v7a\libsmart_camera.so CMakeFiles/smart_camera.dir/src/main/cpp/android_utils.cpp.o CMakeFiles/smart_camera.dir/src/main/cpp/opencv_utils.cpp.o CMakeFiles/smart_camera.dir/src/main/cpp/smart_camera.cpp.o -LD:/APP_android/GitHub/SmartCamera-master/smartcameralib/opencv/lib/armeabi-v7a  -LD:/APP_android/GitHub/SmartCamera-master/smartcameralib/opencv/3rdparty/armeabi-v7a -lopencv_imgproc -lopencv_core C:/Users/GAOKAI/AppData/Local/Android/android-ndk-r21b/toolchains/llvm/prebuilt/windows-x86_64/sysroot/usr/lib/arm-linux-androideabi/libz.a -ltegra_hal -ltbb -lcpufeatures -llog -ljnigraphics -latomic -lm && cd ."
bionic/libc/include/bits/fortify/unistd.h:174: error: undefined reference to '__write_chk'

bionic/libc/include/bits/fortify/unistd.h:174: error: undefined reference to '__write_chk'

clang++: error: linker command failed with exit code 1 (use -v to see invocation)

ninja: build stopped: subcommand failed.

cavanlee, 2020-10-23:
同问
https://github.com/pqpo/SmartCamera/issues/49#issue-719849304

beecool7, 2020-12-21:
我将ndk的版本降低到17就编译成功了，你们可以试一试

gitgeek4dx, 2021-08-18:
不需要降低ndk版本，cmake最后一段，这样改下可以了，ndk21编译正常：

find_library( # Sets the name of the path variable.
              log-lib
              # Specifies the name of the NDK library that
              # you want CMake to locate.
              log )

target_link_libraries( # Specifies the target library.
                       ${TARGET}
                       android
                       # Links the target library to the log library
                       # included in the NDK.
                       ${log-lib}
                       jnigraphics
                       z)

if(${ANDROID_ABI} STREQUAL x86 OR ${ANDROID_ABI} STREQUAL x86_64)
target_link_libraries(${TARGET} opencv_imgproc opencv_core ippiw ippicv ittnotify tbb cpufeatures)
else()
target_link_libraries(${TARGET} opencv_imgproc opencv_core tegra_hal tbb cpufeatures)
endif()